Dasharo/open-source-firmware-validation #36: Multiple automatic tests go into wrong submenu in firmware

johanes2115, 2023-10-10:
Dasharo Security: UEFI Secure Boot test opens TCG Configuration submenu.
Dasharo Security: ME neuter/disable test opens Power Management Options submenu.
Dasharo Security: Network Boot availability test opens One Time Boot submenu.
Tested on Novacustom nv41 ADL.

macpijan, 2023-10-13:
@johanes2115 Please include revision of test environment and logs when reporting problems. Otherwise it would be more challenging to classify and fix them.

johanes2115, 2023-10-13:
@macpijan newest commit where mentioned problems occur: 8a21d7f4301f9323f830d6e8f7827c17af3c6410
[wrong_submenu_tests.zip](https://github.com/Dasharo/open-source-firmware-validation/files/12891962/wrong_submenu_tests.zip)

Psotas, 2023-10-16:
Dasharo Security: UEFI Secure Boot test on VP4670 v2
The keyword `Secure Boot Configuration Submenu` opens the `Driver Health Manager` submenu
[secure-boot_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/12913237/secure-boot_log.zip)

Psotas, 2023-10-25:
Same issue on VP2420:
[reset-to-defaults_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13164219/reset-to-defaults_log.zip)
[bios-lock_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13165294/bios-lock_log.zip)
[smm-bios-write-protection_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13166061/smm-bios-write-protection_log.zip)
[usb-stack_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13166063/usb-stack_log.zip)

Fixes are in preparation:
https://github.com/Dasharo/open-source-firmware-validation/pull/55/

Psotas, 2023-11-06:
[smm-bios-write-protection_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13263809/smm-bios-write-protection_log.zip)
Still some problem with `SMM002.001 SMM BIOS write protection disabling (Ubuntu 22.04)` on VP2420
```
SMM002.001 SMM BIOS write protection disabling (Ubuntu 22.04) :: S... | FAIL |
No match found for 'to boot directly' in 30 seconds. Output:
                /---------------------------------------------\
                |Configuration changed. Reset to apply it Now.|
                |            Press ENTER to reset             |
                \---------------------------------------------/
```

macpijan, 2023-11-07:
@Psotas The other ones has been fixed in the main branch already?

 Have you tried running the test multiple times with the same results?

macpijan, 2023-11-07:
The problems shall be resolved by: https://github.com/Dasharo/open-source-firmware-validation/pull/81 which provides keywords with better reliability in terms of parsing menus

Psotas, 2023-11-13:
@macpijan 

> The other ones has been fixed in the main branch already?

Yes, it works, only with https://github.com/Dasharo/open-source-firmware-validation/issues/36#issuecomment-1794240906 I noticed problems.

> Have you tried running the test multiple times with the same results?

I will check it when the vp4670 v2 platform is available.

closed 2023-12-11 by macpijan