Dasharo/open-source-firmware-validation #42: Dasharo Performance: CPU temperature measure test gets wrong value

johanes2115, 2023-10-10:
Both test cases return:

`56.0Â' cannot be converted to a floating point number: ValueError: could not convert string to float: '56.0Â'`

Tested on Novacustom nv41 ADL.

Psotas, 2023-11-03:
[cpu-temperature_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13249979/cpu-temperature_log.zip)
Same problem on _MSI® PRO Z690-A DDR4_
Dasharo v1.1.3

Psotas, 2024-03-08:
This test needs to be improved and tested on all platforms. Currently, when someone corrects this test for a given platform, it ruins it for another.

VP4650:
```
'100.0°C)
' cannot be converted to a floating point number: ValueError: could not convert string to float: '100.0°C)\n'
```

macpijan, 2024-07-16:
We have something here: https://github.com/Dasharo/open-source-firmware-validation/pull/341

Could be verified on a few different platforms to ensure the proposed solution works fine @BeataZdunczyk

SebastianCzapla, 2025-04-17:
Since last update, CPU temp tests have been updated, and as of today, they work without any issues, closing.

[image]

closed 2025-04-17 by SebastianCzapla