Dasharo/open-source-firmware-validation #43: Dasharo Performance: CPU frequency measure multiple fails

johanes2115, 2023-10-10:
CPF001.001 CPU not stuck on initial frequency (Ubuntu 22.04) returns:

`IndexError: Given index 0 is out of the range 0--1.`

CPF004.001 CPU with load runs on expected frequency (Ubuntu 22.04) fails to authenticate:

`Keyword 'SSHLibrary.Login' failed after retrying 12 times. The last error was: Authentication failed for user 'ubuntu'.`

Tested on Novacustom nv41 ADL.

Psotas, 2023-11-03:
[cpu-frequency_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13249973/cpu-frequency_log.zip)
Same problem on _MSI® PRO Z690-A DDR4_
Dasharo v1.1.3

macpijan, 2024-07-16:
In general, the fix is supposed to be integrated here already: https://github.com/Dasharo/open-source-firmware-validation/pull/361
It can be checked together with previous ticket on CPU temp, on MSI @BeataZdunczyk

philipandag, 2024-08-28:
The errors from this issue are fixed on the current develop branch. Tested on a NV50MB (ubuntu) and V560TND (Ubuntu, Windows).

closed 2024-08-28 by philipandag

philipandag, 2024-08-28:
[ubuntu-with-load_log.zip](https://github.com/user-attachments/files/16778740/ubuntu-with-load_log.zip)
[ubuntu-not-stuck_log.zip](https://github.com/user-attachments/files/16778750/ubuntu-not-stuck_log.zip)
[windows-under-load_log.zip](https://github.com/user-attachments/files/16778824/windows-under-load_log.zip)
The tests were forcibly ended prematurely when it was possible to determine if they work.