Dasharo/open-source-firmware-validation #48: Using tags in tests

Psotas, 2023-10-12:
### The problem you're addressing (if any)

http://robotframework.org/robotframework/latest/RobotFrameworkUserGuide.html#tagging-test-cases

### Describe the solution you'd like

I suggest using tags in tests, this will allow you to trigger/exclude tests based on tags.
Example tags:
[VP4670]
[MSI-Z690]
[dasharo-compatibility]
[dasharo-performance]
[test-reset-to-default]
[test-in-firmware]

Tags can replace things like:
```
Skip If not ${RESET_TO_DEFAULTS_SUPPORT}
Skip If not ${TESTS_IN_FIRMWARE_SUPPORT} RTD011.001 not supported
```

We can prepare a list of tags in the template to only remove individual tags that do not fit, so that no one comes up with tags that no one else knows about

The solution also involves reorganizing the directory structure in the repository into one that will be optimized for creating tests.

I was thinking about something such that, for example, the file name corresponds to the name of a given menu, e.g. `boot-maintanace-manager.robot`, which would make it easier to make sure that all items in the menu are covered in the tests. Unfortunately, for different menus for different devices, this could duplicate tests, such tests could have the tag [duplicated] to know that if they are edited, they should also be edited in other places.

I think tags will help us organize the tests.

pietrushnic, 2023-10-13:
#53 is a possible duplicate of this issue. In general, we see the same problem and need for directory structure improvement.

philipandag, 2024-07-19:
Adressing this in some way here:
https://github.com/Dasharo/open-source-firmware-validation/pull/389

closed 2024-08-14 by BeataZdunczyk