Dasharo/open-source-firmware-validation #49: Add missing variables in platform-name.robot with the value ${TBD}

Psotas, 2023-10-12:
### Describe the solution you'd like

Currently, platform-name.robot files have a different number of variable names, when adding variables we should strive (automate(?)) completion so that each file has the same number of variables. This will help us avoid that something has not been configured/added for the platform.

Unset/unconfigured variables should be set to ${TBD}.
The value of ${TBD} should be some random string (generated each time(?).
We are currently using ${EMPTY}, which means that some of the tests are false positive. For example, waiting to read a prompt in telnet which will return TRUE with the value of ${EMPTY}.

macpijan, 2024-07-16:
@BeataZdunczyk  We can in fact consider changing the empty strings from https://github.com/Dasharo/open-source-firmware-validation/blob/develop/platform-configs/include/default.robot#L20 with some magic string to get rid of false positives.

We may need to check if we rely on Skipping some tests based on the string content.

philipandag, 2024-07-19:
Addressing the issue here:
https://github.com/Dasharo/open-source-firmware-validation/pull/393

philipandag, 2024-07-19:
> Currently, platform-name.robot files have a different number of variable names, when adding variables we should strive > (automate(?)) completion so that each file has the same number of variables.

I am a bit concerned about that. Would not that imply that the `include/default.robot` file would not be needed anymore? Would not that cause lots of redundant variable defines?

mkopec, 2024-07-24:
The issue was created before `default.robot` was created, back then every platform re-defined every single variable and it was hard to maintain.

Now, the part of the issue we want to address here, is adding new platforms automatically. A script could start with an empty config that just includes `default.robot`, and iterate through every var in `default.robot` to check if the value is correct for a given platform. If it isn't, it should set it correctly in the platform config file.

philipandag, 2024-07-24:
> check if the value is correct for a given platform. If it isn't, it should set it correctly in the platform config file.

@mkopec So we do not want to repeat any values that are set in `default.robot` if not needed, but how could we automatically check if the values are correct for the platform? 

Does that mean that the platform-dependent common files should include every variable that might get used by any of the platforms it is used in and be used as a template for those configs, not a common include?

mkopec, 2024-08-20:
https://github.com/Dasharo/open-source-firmware-validation/pull/393 merged, closing

closed 2024-08-20 by mkopec