Dasharo/open-source-firmware-validation #51: Small steps keywords

Psotas, 2023-10-12:
### Describe the solution you'd like

Keywords should be as simple as possible, remember that Robot Framework is used to easily read what a given step does. We should look at it as if we were writing manual instructions and the keywords should correspond to the next steps of such instructions.

This will allow for easier reuse of such keywords, they will not be strongly related to a given case, which may ultimately help to optimize the number of keywords.

https://github.com/Dasharo/open-source-firmware-validation/pull/45/files#diff-6ad7c5fb97cb1099c7ecd42f15dde3caf06d38b9fe5244afb96394454b6d6bfaL40-R80
`Enter Dasharo System Features Submenu`
`Enter Dasharo System Features Submenu Snapshot`
these are examples of keywords that, due to their extensiveness, are not clear and universal.
That's why I created keywords
`Enter Dasharo Submenu`
`Enter Dasharo Submenu Snapshot`
they are less complicated, and therefore more readable, and more universal.

When reading the test, we must immediately know how to perform it manually. In the future, this will help in automatically generating manual instructions.

https://github.com/Dasharo/open-source-firmware-validation/pull/45/files#diff-3c157e9d4762250f95224f014fa3acb1f979e1aca0c9fe3808a4156f980d7cc7R69-R73
For example, I took the liberty of creating a keyword that calls another keyword to make it easier to understand why such a step was included here. This will also help in generating documentation.

macpijan, 2023-10-15:
> https://github.com/Dasharo/open-source-firmware-validation/pull/45/files#diff-3c157e9d4762250f95224f014fa3acb1f979e1aca0c9fe3808a4156f980d7cc7R69-R73
For example, I took the liberty of creating a keyword that calls another keyword to make it easier to understand why such a step was included here. This will also help in generating documentation.

This does not make much sense. The `Power On` keyword simply should never be used here. `Save Changes and Reset` should be used, as here: https://github.com/Dasharo/open-source-firmware-validation/pull/55/files#diff-3c157e9d4762250f95224f014fa3acb1f979e1aca0c9fe3808a4156f980d7cc7R39

macpijan, 2023-10-15:
This should be simple enough to reproduce manually: https://github.com/Dasharo/open-source-firmware-validation/pull/55/files#diff-3c157e9d4762250f95224f014fa3acb1f979e1aca0c9fe3808a4156f980d7cc7R28 
Let me know what do you think about that.

Psotas, 2023-10-16:
@macpijan

> Save Changes and Reset should be used

You're right. I followed the example of `reset-to-defaults.robot`, which unfortunately does not contain such a keyword.

> Let me know what do you think about that.

The test is understandable what it does step by step.

`Enter Setup Menu Tianocore And Return Construction`
`Enter Submenu From Snapshot And Return Construction` 
This is a great example that a good keyword name can replace keyword documentation.