Dasharo/open-source-firmware-validation #53: Get rid of deprecated division of thests into Dasharo modules

pietrushnic, 2023-10-12:
Dasharo modules concept was never correctly described nor introduced in a consistent way. Despite that, it is widely used. We should educate ourselves on the practices of test organization and apply that methodology, especially if some or all of those tests will drive product development.

My expectation is to stop using modules: 
* `dasharo-compatibility`,
* `dasharo-security`,
* `dasharo-performance`,
* `dasharo-stability

And creation of structure for test division which better express segments of functionalities which Dasharo support.

macpijan, 2023-10-13:
Do you have any concrete ideas in mind how it can be improved?

macpijan, 2023-10-13:
It originates from here: https://docs.dasharo.com/value-prop/compatibility/ and even before

SebastianCzapla, 2024-11-14:
Is this issue still relevant at this time? Currently, there is a bit of chaos within the test directories. Similar subjects are split into multiple directories, or even files within them, for example, TPM tests are split into dasharo-security/tpm-support.robot, dasharo-security/tpm2-commands.robot, dasharo-stability/tpm-detect.robot dasharo-stability/tpm-detect-heads.robot