Dasharo/open-source-firmware-validation #54: requirements.txt conflict robotframework-sherlock and mkdocs

pietrushnic, 2023-10-12:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
mkdocs 1.5.2 requires pathspec>=0.11.1, but you have pathspec 0.9.0 which is incompatible.
```

I used the same venv for both mkdocs and osfv. Using new clean venv solves issue for me.

pietrushnic, 2023-10-13:
@macpijan I have no idea how to classify this. I wasted some time before I realized there was conflict, and I assume it can happen also to other users. Separate venv for each repo would eliminate the problem but may lead to redundancy and storage waste, so the question is, what is our KPI here? My rough guess is we would like to deliver a reliable environment. Also, I wonder if we shouldn't get back to the docker image instead of using venv and system-native python. It would eliminate this problem also.

macpijan, 2023-10-16:
> Separate venv for each repo would eliminate the problem

Yes, and this is what README suggests and is sane to do.
TBH this is quite unusual (and a bad practice in general) to use the same venv for multiple projects at the same time (with different requirements.txt files).

> what is our KPI here?

That following documented steps produces a working environment?

> lead to redundancy and storage waste,

> Also, I wonder if we shouldn't get back to the docker image instead of using venv and system-native python. It would eliminate this problem also.

So 100MB venv is redundancy and storage waste, but 300+MB container is not? In the end, container has the same as venv inside + more.

We might go back to docker. We would need to add some wrapper scripts so it would be easier to use.
I must admit, that using venv is a bit easier and more convenient to use, especially if we need to modify the venv, not just execute scripts.

I think it might be easier to keep in sync by placing Dockerfile in this repo (so the same requirements.txt can be resued), rather than having to sync another repo: https://github.com/3mdeb/rf-docker

pietrushnic, 2023-10-16:
> TBH this is quite unusual (and a bad practice in general) to use the same venv for multiple projects at the same time (with different requirements.txt files).

Correct, this is bad practice. I will close this issue since it is caused by having too much in one venv.

> So 100MB venv is redundancy and storage waste, but 300+MB container is not? In the end, container has the same as venv inside + more.

The question is what is more reliable venv or docker.

> We might go back to docker. We would need to add some wrapper scripts so it would be easier to use.
I must admit, that using venv is a bit easier and more convenient to use, especially if we need to modify the venv, not just execute scripts.

It really depend if you create this project for users or for developers.

> I think it might be easier to keep in sync by placing Dockerfile in this repo (so the same requirements.txt can be resued), rather than having to sync another repo: https://github.com/3mdeb/rf-docker

Agree, maybe even CI for that?

closed 2023-10-16 by pietrushnic

macpijan, 2023-10-16:
https://github.com/Dasharo/open-source-firmware-validation/issues/60