Dasharo/open-source-firmware-validation #56: Missing Linux configuration on the platform ('flashrom' not found)

Psotas, 2023-10-16:
Tested on VP4670 v2.

`KEYWORD ${out_flashrom} = keywords . Execute Command In Terminal flashrom -p internal`
`Command 'flashrom' not found, but can be installed with:
apt install flashrom`

Should we have some sort of auto-check for system configuration, or auto-installers for missing system component errors?
[missing-flashrom-logs.zip](https://github.com/Dasharo/open-source-firmware-validation/files/12913319/missing-flashrom-logs.zip)

macpijan, 2023-10-16:
@Psotas The `Get Flashrom From Cloud` already checks for that. Simply use this keyword in keywords using flashrom for now, until we have better solution.

macpijan, 2023-10-16:
**do not** install flashrom from apt, in most cases this is not enough.

macpijan, 2024-07-16:
Here we have some ansible proposal: https://github.com/Dasharo/open-source-firmware-validation/pull/212 @BeataZdunczyk 
If we finalize integration of these two big MR, we can apply ansible approach to the rest of the tests as well @BeataZdunczyk  (if it works well for us).

SebastianCzapla, 2025-04-25:
Fixed in: https://github.com/Dasharo/open-source-firmware-validation/pull/742

closed 2025-04-25 by SebastianCzapla