Dasharo/open-source-firmware-validation #57: Remove the ${Y} variable in the `Setup Intel ME Mode` keyword

Psotas, 2023-10-16:
We use the ${Y} variable in the `Setup Intel ME Mode` keyword
We don't have the ${Y} variable defined anywhere.
Can be replaced by the keyword `Save Changes and Reset`

macpijan, 2023-10-16:
This report provides very limited value. At the very least, please provide revision of repo on which you are talking about. Better, provide a link to the problem. Ideally, just submit a PR with a fix instead of issue, as it sounds like a straghtforward one...

Psotas, 2023-11-15:
Issue solved by https://github.com/Dasharo/open-source-firmware-validation/commit/3deec03fa2d8a87d6e651c3e25dd9392df23d68b (The keyword `Setup Intel ME Mode` is no longer used.)

closed 2023-11-15 by Psotas