Dasharo/open-source-firmware-validation #58: Sometimes `Power On` just starts the operating system's shutdown procedure.

Psotas, 2023-10-16:
Sometimes, the `Power On' keyword does not restart the platform, but causes it to be turned off via the operating system shutdown procedure.
[measured-boot_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/12913731/measured-boot_log.zip)

philipandag, 2024-08-20:
We could define a new tag for doing this second step of basic tests.

philipandag, 2024-09-12:
The issue doesn't reappear on the current osfv.
[ubuntu_boot_freeze_log.zip](https://github.com/user-attachments/files/16974035/ubuntu_boot_freeze_log.zip)
Note that the logs show another unrelated problem when booting to Ubuntu which is described here https://github.com/Dasharo/open-source-firmware-validation/issues/498

closed 2024-09-12 by philipandag