Dasharo/open-source-firmware-validation #61: Missing `Boot From USB` keyword, usb-boot.robot tests do not work.

Psotas, 2023-10-16:
Missing `Boot From USB` keyword, usb-boot.robot tests do not work.
[usb-boot_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/12915501/usb-boot_log.zip)

BeataZdunczyk, 2024-07-16:
working on here:  https://github.com/Dasharo/open-source-firmware-validation/pull/351

m-iwanicki, 2024-07-16:
@BeataZdunczyk It's duplicate of https://github.com/Dasharo/open-source-firmware-validation/issues/279 and possibly https://github.com/Dasharo/open-source-firmware-validation/issues/143
which should be fixed when @PLangowski PR https://github.com/Dasharo/open-source-firmware-validation/pull/312 is merged.

SebastianCzapla, 2025-04-17:
Fixed in https://github.com/Dasharo/open-source-firmware-validation/pull/659

Example pass in https://github.com/Dasharo/open-source-firmware-validation/pull/659#issuecomment-2596277720

closed 2025-04-17 by SebastianCzapla