Dasharo/open-source-firmware-validation #67: Refactor `Flash Firmware` Keyword to prepare the platform for internal flashing ex. for `verified-boot.robot` tests

Psotas, 2023-10-25:
### The problem you're addressing (if any)

`Flash BIOS Region Via Internal Programmer` does not work. because some of the memory is read-only, so it returns the message `Make sure that SPI locks are disabled prior flashing internally`

### Describe the solution you'd like

Add a keyword to prepare the platform for integral flashing.
Maybe refactor the entire `Flash Firmware` keyword?

macpijan, 2023-10-25:
@Psotas 

> Flash BIOS Region Via Internal Programmer does not work.

Have you tried disabling locks first manually?

The keyword is detecting the fact that locks are enabled and printing some reasonable information. I remember it's state where it would use the flashrom either way, causing brick most likely.

So I would not say it does not work, but can be further improved/automated, by disabling the locks if necessary.

Psotas, 2023-10-26:
`Pass` after manually disabling locks.

This is enhancement issue, not bug issue. 
Goal is make test full automated.

Can also add appropriate actions if flashing fails. e.g., halt further test execution to allow for repairs with an integral programmer.

closed 2023-12-05 by macpijan