Dasharo/open-source-firmware-validation #68: `secure-boot.robot`: refactor test to not use `Upload Required Images` keyword.

Psotas, 2023-10-25:
### The problem you're addressing (if any)

Instead of using PiKVM to copy files to USB, we can simply use files that we manually placed on it.

### Describe the solution you'd like

Just use file from USB. Not every platform has PiKVM.

Psotas, 2023-11-17:
https://github.com/Dasharo/open-source-firmware-validation/pull/118#discussion_r1397115284

macpijan, 2024-07-16:
@BeataZdunczyk Currently, we do not run these extra Secure Boot tests on non-PiKVM platforms (such as Protectli).
That could be done if we have USB with such files as suggested.

The goal here would be:
- check if in the current state of tests we can run full SB suite on Protectli,
- see what's missing
- adjust the test so the files can be provided non only via PiKVM, bu also from USB stick

pietrushnic, 2024-07-17:
@macpijan, all UEFI Secure Boot tests are aligned with my training plans (Arch4221 and TC3211), so if there is any way I can help, I would be glad to assume it would be done on Odroid-H4+. Feel free to cc me wherever such a connection exists.

SebastianCzapla, 2025-04-25:
Fixed in https://github.com/Dasharo/open-source-firmware-validation/pull/725

closed 2025-04-25 by SebastianCzapla