Dasharo/open-source-firmware-validation #74: `MBO001.001 Measured Boot support (Ubuntu 20.04)` test fail

Psotas, 2023-10-27:
### Device

MSI® PRO Z690-A DDR4 and VP2420

### RTE version

MSI v1.1.3 and VP2420 v1.2.0

### Brief summary

No found matches `^\\?TPM: Digest of .* to PCR \\d+ measured$`  in `cbmem -1 | grep PCR`

### How to reproduce

Run test `MBO001.001 Measured Boot support (Ubuntu 20.04)`

### Expected behavior

Test Pass when found correct value

### Actual behavior

Test fail, because `^\\?TPM: Digest of .* to PCR \\d+ measured$` don't found matches in `cbmem -1 | grep PCR`

### Link to screenshots or logs

[measured-boot_log_MSI-690-A_DDR4.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13188044/measured-boot_log_MSI-690-A_DDR4.zip)
[measured-boot_log_VP2420.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13188057/measured-boot_log_VP2420.zip)

### Additional context

When the test execute `cbmem -1 | grep PCR` we have returns lines.
When trying to use  ` Get Lines Matching Regexp` `^\\?TPM: Digest of .* to PCR \\d+ measured$` doesn't find any matches.

macpijan, 2023-11-01:
@mkopec @krystian-hebel 

I assume these logs in coreboot may change with version, and due to other reasons. We already have some examples:
- https://docs.dasharo.com/unified-test-documentation/dasharo-security/203-measured-boot/#mbo001001-measured-boot-support-ubuntu-2204
- https://docs.dasharo.com/variants/asus_kgpe_d16/tpm-mboot/#measured-boot-behaviour

Can we define a better way of measured boot verification?

The end result is that measurements are stored in a set of PCRs. We know which one should be extended, correct?

As a first step, we could check if these are non-zero (or non-default value)?

As a second, if they match the expected PCRs for the given release, assummng that we could pre-calculate that?

pietrushnic, 2023-11-02:
There is definitely a need for serious measured boot testing. In an ideal world, we would have SBOM published in some reasonable format, and thanks to its value, we could predict the values of PCRs. This should be prototyped in QEMU with swtpm first. Such work should also help validate Heads.

closed 2023-11-16 by macpijan