Dasharo/open-source-firmware-validation #77: `Flash Firmware` keyword does not work for MSI® PRO Z690-A DDR4 platform

Psotas, 2023-10-30:
### Device

MSI® PRO Z690-A DDR4

### RTE version

RTE Rev v1.1.0 

### Affected component(s) or functionality

The entire `Flash Firmware` keyword would probably need to be reviewed and refactored.

### Brief summary

Fail `Flash Firmware` during external flashing of the platform MSI® PRO Z690-A DDR4

### How to reproduce

Run `verified-boot.robot` test

### Expected behavior

Pass `Flash Firmware` 

### Actual behavior

FAIL `Flash Firmware` 

### Link to screenshots or logs

[verified-boot_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13202974/verified-boot_log.zip)

macpijan, 2023-11-01:
@Psotas Have you actually inspected the flashrom log from attached logs?

```
  | Return: ['flashrom v1.2 on Linux 5.4.69 (armv7l)\nflashrom is free software, get the source code at https://flashrom.org\n\nUsing region: "bios".\nUsing clock_gettime for delay loops (clk_id: 1, resolution: 1ns).\nNo EEPROM/flash device found.\nNote: flashrom can never write if the flash chip isn\'t found automatically.',  1]
-- | --
```

It cannot find chipset, meaning incorrect flashrom version was used.

macpijan, 2023-11-01:
The `Get flashrom` keyword downloads following one: `flashrom 28a8723 on Linux 6.4.15-200.fc38.x86_64 (x86_64)
` - note the `28a8723` revision.
You may need to remove the flashrom from apt if you have installed it.

macpijan, 2023-11-01:
But if you want to flash in test, there will be another problem - the serial will be disabled by default, so the test will not continue anymore.

Psotas, 2023-11-02:
[verified-boot_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13241073/verified-boot_log.zip)
I ran a retest, the platform was successfully flashed.
I'm closing the issue.
New issue `verified-boot.robot` test problem: https://github.com/Dasharo/open-source-firmware-validation/issues/85

closed 2023-11-03 by Psotas