Dasharo/open-source-firmware-validation #78: Sometimes Windows enters recovery mode instead of starting normally.

Psotas, 2023-10-30:
### Device

MSI® PRO Z690-A DDR4

### RTE version

Dasharo v1.1.3

### Brief summary

Sometimes Windows enters recovery mode instead of starting normally.

### How reproducible

3%

### How to reproduce

Power restart, without "properly closing windows"

### Expected behavior

If the SSH connection fails, we should reset the power, e.g. 3 times.

### Actual behavior

Tests fail. SSH connection fails because the system does not boot.

### Link to screenshots or logs

[image]

Psotas, 2023-10-30:
Potential solutions:
1. Wait for full turn off platform.
2. Force blue screen and then reboot. Maybe this will turn off windows at a better time than after starting the shutdown procedure after pressing the power button.

macpijan, 2023-10-30:
> If the Login To Windows Via SSH fails, we should reset the power, e.g. 3 times.

This sounds more like a workaround. The proper fix would be to shutdown Windows gracefully before turning off power.

I definitely opt for option number 1.

macpijan, 2023-11-07:
So far, we will stick with the workaround implemented in: https://github.com/Dasharo/open-source-firmware-validation/pull/84

macpijan, 2024-07-16:
@BeataZdunczyk This is something that causes some problems during testing, that can break Windows if no powering off gracefully.

That would require some more thought into it though, how to ensure we always try to poweroff gracefully, prior to removing power.