Dasharo/open-source-firmware-validation #82: Parse coreboot config to check for expected features

macpijan, 2023-11-01:
### The problem you're addressing (if any)

We could check the coreboot cofnig from binary to see which features should be expected to enabled, or what the default values should be.

### Additional context

See discussion here: https://github.com/Dasharo/open-source-firmware-validation/pull/29/#discussion_r1352083455

macpijan, 2024-07-16:
@BeataZdunczyk Maybe a bit more complex, but could be another step in generating configs automatically - to detect which features should be enabled, and which test matrix should be applied for the given hardware.

So related to feature/test matrix topics as well.

filipleple, 2024-09-12:
Base mechanism with some example options implemented in https://github.com/Dasharo/open-source-firmware-validation/pull/392. It works, the list of available `option:test_flag` pairs would ideally be constantly extended.

closed 2024-09-12 by filipleple