Dasharo/open-source-firmware-validation #87: `USB001.003 USB devices detected by OS (Windows 10)` test FAIL

Psotas, 2023-11-03:
### Device

MSI® PRO Z690-A DDR4

### RTE version

Dasharo v1.1.3

### Brief summary

`USB003.00X Upload 1GB file on USB storage` test FAIL

### How to reproduce

Run `usb-hid-and-msc-support.robot`

### Expected behavior

Test PASS

### Actual behavior

Test FAIL

### Link to screenshots or logs

[usb-hid-and-msc-support_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13248067/usb-hid-and-msc-support_log.zip)

m-iwanicki, 2024-07-04:
Failing `USB001.003 USB devices detected by OS (Windows 10)` was fixed earlier here: https://github.com/Dasharo/open-source-firmware-validation/commit/02f467227bbd935d9677811bcec28677f985fb02

Failing `USB003.001` should be fixed by https://github.com/Dasharo/open-source-firmware-validation/pull/351

Tested on Protectli-VP2420:

```
robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.221 -v device_ip:192.168.10.161 -v config:protectli-vp2420 dasharo-compatibility/usb-hid-and-msc-support.robot
==============================================================================
Usb-Hid-And-Msc-Support
==============================================================================
USB001.001 USB devices detected in FW :: Check whether USB devices... | PASS |
------------------------------------------------------------------------------
USB001.002 USB devices detected by OS (Ubuntu) :: Check whether th... ......
Checking if usbutils is installed...

Package usbutils is installed
USB001.002 USB devices detected by OS (Ubuntu) :: Check whether th... | PASS |
------------------------------------------------------------------------------
USB001.003 USB devices detected by OS (Windows) :: Check whether t... | PASS |
------------------------------------------------------------------------------
USB002.001 USB keyboard detected in FW :: Check whether the extern... | SKIP |
USB002.001 not supported
------------------------------------------------------------------------------
USB002.002 USB keyboard in OS (Ubuntu) :: Check whether the extern... | SKIP |
USB002.002 not supported
------------------------------------------------------------------------------
USB002.003 USB keyboard in OS (Windows) :: Check whether the exter... | SKIP |
USB002.003 not supported
------------------------------------------------------------------------------
USB003.001 Upload 1GB file on USB storage (Ubuntu) :: Check whethe... | PASS |
------------------------------------------------------------------------------
USB003.002 Upload 1GB file on USB storage (Windows) :: Check wheth... | FAIL |
No match found for 'to boot directly' in 30 seconds. Output:
------------------------------------------------------------------------------
Usb-Hid-And-Msc-Support                                               | FAIL |
8 tests, 4 passed, 1 failed, 3 skipped
==============================================================================
```

`USB003.002` probably failed to `Power On` correctly this try, second one:

```
vp2420 -t "USB003.002*" dasharo-compatibility/usb-hid-and-msc-support.robot
==============================================================================
Usb-Hid-And-Msc-Support
==============================================================================
USB003.002 Upload 1GB file on USB storage (Windows) :: Check wheth... | PASS |
------------------------------------------------------------------------------
Usb-Hid-And-Msc-Support                                               | PASS |
1 test, 1 passed, 0 failed
==============================================================================
```

logs:
[logs.tar.gz](https://github.com/user-attachments/files/16100509/logs.tar.gz)

SebastianCzapla, 2025-04-24:
https://github.com/Dasharo/open-source-firmware-validation/pull/810

Issue was present on develop as of 24.04.2025, so I wrote a fix above.

SebastianCzapla, 2025-04-29:
Closing as https://github.com/Dasharo/open-source-firmware-validation/pull/810 was merged