Dasharo/open-source-firmware-validation #99: Improve reliability of Exit From Current Menu keyword

macpijan, 2023-11-10:
https://github.com/Dasharo/open-source-firmware-validation/pull/95#issuecomment-1805561335

macpijan, 2023-11-10:
`for i in {1..20}; do robot -t "Make Sure That Keys Are Provisioned"  -L TRACE -v config:qemu -v rte_ip:127.0.0.1 -v snipeit:no self-tests/secure-boot.robot; done` - gives 3/20 failures for both 1s or 2s delays.

macpijan, 2023-11-10:
0.1s: 20/20 fails

macpijan, 2024-07-16:
This may make some sense still to investigate and add test for stress testing this kwd (as we do with `Power On` one) @BeataZdunczyk .

Haven't you had some problems with this kwd @PLangowski ? Did you have some test for it?

PLangowski, 2024-07-16:
I had problems with `Reenter menu` https://github.com/Dasharo/open-source-firmware-validation/pull/329