Dasharo/open-source-firmware-validation #115: Tests in boot-measure.robot will not fail with long, irregular boot times.

Psotas, 2023-11-16:
### The problem you're addressing (if any)

Currently, tests will pass as soon as the OS works. To detect irregularities during startup, you need to look at the values manually.

### Describe the solution you'd like

Ideas:
1. Compare the times between boot times to see if they are too different.
2. Compare the average time with the expected time (variable for the platform?)
3. Compare whether the time between boot types is not too different.

### Where is the value to a user, and who might that user be?

It would be possible to detect when the time is too long compared to expectations.

macpijan, 2024-07-16:
@BeataZdunczyk We can define some expected time in platform config.

The challenge would be here to define empirically this value for each platform.
We can start with some big number everywhere. Then, we can do that with each release testing, to avoid extra tasks here.

philipandag, 2024-07-22:
My idea would be to make the `boot-time-measure.robot` suite measure, aside from the average boot time, the standard deviation, the lowest and the highest boot time which could all be compared against some, like @macpijan suggests, empirically tested target values. 

All these values would be logged to the console after the test and the test could fail if, for example, the standard deviation is too high or the longest boot time was too long. 

Please share your thoughts about that.

philipandag, 2024-07-22:
PR: https://github.com/Dasharo/open-source-firmware-validation/pull/399

philipandag, 2024-08-12:
The PR was merged. Boot time measure tests now also measure min, max, average and the standard deviation of the boot times and compare it against configurable values allowing for controlling the boot times more precisely. The boot time statistics acceptable values still need to be defined more precisely as they are now set to some arbitrary high values as a placeholder. Selecting the values for every platform would probably be a seperate task.

philipandag, 2024-08-12:
https://github.com/Dasharo/open-source-firmware-validation/issues/450

closed 2024-08-12 by philipandag