Dasharo/open-source-firmware-validation #119: Sometime `Enter Setup Menu Tianocore And Return Construction` return unexpected menu construction.

Psotas, 2023-11-17:
### Device

MSI® PRO Z690-A DDR5

### RTE version

Dasharo v1.1.3

### Brief summary

Sometime `Enter Setup Menu Tianocore And Return Construction` return unexpected menu construction.

### How reproducible

<5%

### How to reproduce

e.g. run secure-boot.robot

### Expected behavior

Return: [
```
'Select Language            <Standard English>',
 '> User Password Management',
 '> Device Manager',
 '> Dasharo System Features',
 '> One Time Boot',
 '> Boot Maintenance Manager',
 'Continue',
 'Reset'
 ```
 ]

### Actual behavior

MSI® PRO Z690-A DDR5:
Return: [
```
'Select Language            <Standard English>         This is the option',
 'one adjusts to change',
 '> User Password Management                              the language for the',
 '> Device Manager                                        current system',
 '> Dasharo System Features',
 '> One Time Boot',
 '> Boot Maintenance Manager',
 'Continue',
 'Reset'
 ```
 ]
 
VP4630 (It looks the same for the 3 caught errors):
Return: [
```
'Select Language <Standard English> This is the option',
 'one adjusts to change',
 '> Device Manager the language for the',
 '> Dasharo System Features current system',
 '> One Time Boot',
 '> Boot Maintenance Manager',
 'Continue',
 'Reset'
 ```
 ]

### Link to screenshots or logs

[secure-boot_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13391891/secure-boot_log.zip)
[image]

macpijan, 2023-11-28:
@Psotas Can you identify when it happens? E.g. can you provide the exact command to be run in the loop 20 times and we can see the reproduction rate?

The uefi-shell tests would do?

Psotas, 2023-11-28:
@macpijan 
I am currently working on reworking the `Platform-Suspend-And-Resume` tests on novacustom and this happens from time to time. I am unable to reproduce this problem now.

From what I noticed, the structure of the incorrectly read menu is always the same for a given platform.

> The uefi-shell tests would do?

This test do not enter to bios menu. Maybe this command will be effective. 
```
robot -L TRACE -v rte_ip:192.168.4.180 -v config:novacustom-nv41pz -v snipeit:no -t MNE001.* dasharo-security/me-neuter.robot
```
I suspect this may occur when we run multiple tests, no than one single test.

PLangowski, 2023-12-01:
I have encountered the same error. Below is the test log.

[secure-boot_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13527210/secure-boot_log.zip)

macpijan, 2024-07-16:
@BeataZdunczyk another issue that first needs that we write some stress test for this keyword to isolate the problem and see what kind of reproducibility we have here

philipandag, 2024-08-12:
The issue could not be replicated in qemu nor on msi-z690 ddr5 in 200 tries each. Here are the logs:
[unexpected-construction.zip](https://github.com/user-attachments/files/16578834/unexpected-construction.zip)

closed 2024-08-20 by BeataZdunczyk