Dasharo/open-source-firmware-validation #121: Do not rely on cloud.3mdeb.com to download test data

macpijan, 2023-11-20:
### The problem you're addressing (if any)

We should not rely on cloud.3mdeb.com to download test data:
- it increases cloud.3mdeb.com usage, which is not necessary,
- in case of outage, tests would not work,
- cannot be contributed to by external contributors.

### Describe the solution you'd like

Use a repo independent of 3mdeb internal infrastructure, which we can still control and approve external contributions. Most likely in a form of a github repository.

macpijan, 2024-07-16:
@BeataZdunczyk We need to update the status of where cloud is used.
Then, we need to propose alternative solution.
Even creating a git repo as a submodule with test files, can be a viable option.

macpijan, 2024-09-02:
@wiktormowinski Let's start with updating a list of required components.

Maybe the new repo makes sesnse, as it was one of the ideas in linked discussion. This repo then would be a submodule of OSFV.

Maybe we can enable LFS in this repo if there will be many binary files there.

wiktormowinski, 2024-09-02:
@macpijan Updated list looks like that 
keywords.robot:
```
Download File    https://cloud.3mdeb.com/index.php/s/C6LJMi4bWz3wzR9/download    ${cbmem_path}
Download File    https://cloud.3mdeb.com/index.php/s/fsPNM8SpDjATMrW/download    ${flashrom_path}
Download File    https://cloud.3mdeb.com/index.php/s/ScCf8XFLZYWBE25/download    ${cbfstool_path}
```
dasharo-compatibility/dcu.robot:
```
    Download To Host Cache
    ...    logo.bmp
    ...    https://cloud.3mdeb.com/index.php/s/rsjCdz4wSNesLio/download
    ...    6e5a6722955e4f78d947654630f27ff833703fbc04776ffed963c96617f6bb2a
``` 
in lib/esp-scannning-lib.robot
```
 ${DISK_IMAGE_URL}=              https://cloud.3mdeb.com/index.php/s/BwLyjHT9fRncXMY/download/image.img
```
lib/secure-boot-lib.robot
```
${GOOD_KEYS_URL}=           https://cloud.3mdeb.com/index.php/s/SjM5dQGji4XDAni/download/good_keys.img
${NOT_SIGNED_URL}=          https://cloud.3mdeb.com/index.php/s/zmJXxGG4piGB2Me/download/not_signed.img
${BAD_KEYS_URL}=            https://cloud.3mdeb.com/index.php/s/BJPbSqRH6NdbRym/download/bad_keys.img
${BAD_FORMAT_URL}=          https://cloud.3mdeb.com/index.php/s/AsBnATiHTZQ6jae/download/bad_format.img
```

wiktormowinski, 2024-09-02:
sure, any requests regarding the name of this new submodule (and other such parameters)?

macpijan, 2024-09-02:
No. osfv-test-data?

macpijan, 2025-03-14:
Merged in: https://github.com/Dasharo/open-source-firmware-validation/pull/667

closed 2025-03-14 by macpijan