Dasharo/open-source-firmware-validation #129: `Set Option State` do not work for `Platform sleep type`

Psotas, 2023-11-29:
### Device

novacustom nv41pz

### RTE version

v1.7.1

### Brief summary

`Set Option State` do not work for `Platform sleep type`

### How to reproduce

run test on PR https://github.com/Dasharo/open-source-firmware-validation/pull/128
```
robot -L TRACE -v rte_ip:192.168.4.180 -v config:novacustom-nv41pz -v snipeit:no dasharo-compatibility/platform-suspend-and-resume.robot
```

### Expected behavior

-

### Actual behavior

-

### Link to screenshots or logs

[image]
[log-Power-Management-Options.zip](https://github.com/Dasharo/open-source-firmware-validation/files/13489178/log-Power-Management-Options.zip)
```Output:

                         /---------------------------\
                         | Suspend to Idle (S0ix)    |
                         | Suspend to RAM (S3)       |

  ^v=Move Highlight       <Enter>=Complete Entry    Esc=Exit Entry
  LCtrl+LAlt+F12=Save screenshot
```

### Additional context

The error is probably on Dasharo's side, in the way it sends data by serial.
Should we look for workaround in tests or maybe report it in dasharo issues?

macpijan, 2023-12-06:
@Psotas The workaround is always the same as we have already done before for laptops. Eg. move up and down, I suppose.

macpijan, 2023-12-06:
> Should we look for workaround in tests or maybe report it in dasharo issues?

We need tracking for that in dasharo-issues as well, otherwise we will not go back to this problem ever.

macpijan, 2023-12-11:
@Psotas I assume that with the w/a it already works. For the original problem, we have another issue linked above.

closed 2023-12-11 by macpijan