Dasharo/open-source-firmware-validation #138: `Get Submenu Construction` sometimes returns unexpected construction

PLangowski, 2023-12-08:
### Device

MSI® PRO Z690-A DDR5

### RTE version

Dasharo v1.1.3

### Brief summary

`Get Submenu Construction` sometimes returns unexpected construction

### How to reproduce

Running `secure-boot.robot`

### Expected behavior

Return: ['> TCG Drive Management',
 '> Driver Health Manager',
 '> RAM Disk Configuration',
 '> Secure Boot Configuration',
 '> HDD Security Configuration',
 '> TCG2 Configuration',
 'Press ESC to exit.']

### Actual behavior

Return: ['Devices List                                          Select to manage',
 '> TCG Drive Management',
 '> Driver Health Manager',
 '> RAM Disk Configuration',
 '> Secure Boot Configuration',
 '> HDD Security Configuration',
 '> TCG2 Configuration',
 'Press ESC to exit.']

### Link to screenshots or logs

[image]

### Additional context

This issue is similar to https://github.com/Dasharo/open-source-firmware-validation/issues/119

macpijan, 2024-07-16:
@BeataZdunczyk Another one that may need adding some stress/stability test to narrow down the problem and monitor reliability of framework.

I am starting adding such label.

JanPrusinowski, 2024-07-30:
This issue did not replicate on qemu: 
[log.zip](https://github.com/user-attachments/files/16424154/log.zip)
I have run a test which reads menu construction 50 times and it didn't fail even once. I am going to test it on the MSI® PRO Z690-A DDR5 to see if the error still occurs.

JanPrusinowski, 2024-07-30:
I have managed to replicate this issue.
[log.zip](https://github.com/user-attachments/files/16427144/log.zip)
[image]

JanPrusinowski, 2024-07-31:
I have added a delay in Get Menu Construction and it lokks like it fixed the issue.
200 repeats and issue didnt replicate even once:
[log.zip](https://github.com/user-attachments/files/16437073/log.zip)

macpijan, 2024-08-05:
Thanks @JanPrusinowski I have merged it, let's see if that solves it in the long run as well.

closed 2024-08-05 by macpijan