Dasharo/open-source-firmware-validation #141: dts-lib: rework flashing method

filipleple, 2023-12-08:
**Test environment version**
* Branch: 
* Commit: 

**Dasharo version (if applicable)**

**Dasharo variant (if applicable)**

**Affected component(s) or functionality (if applicable)**

flashing keywords in [dts-lib](https://github.com/Dasharo/open-source-firmware-validation/blob/main/lib/dts-lib.robot)

**Brief summary**

In [`lib/dts-lib.robot`](https://github.com/Dasharo/open-source-firmware-validation/blob/main/lib/dts-lib.robot), there is a keyword called `Flash Firmware In DTS` for flashing that forces flashing with the internal programmer. It needs to be replaced with the more generic keyword that we already have developed in [`keys-and-keywords/flashrom.robot`](https://github.com/Dasharo/open-source-firmware-validation/commit/93c488b11358bc22c6b91c16c946f33c69781578#diff-f0872f96b8d235621587f018ef3d8e5bd51ba8e092be1929279b2436f62ec83e).

**Additional context**

macpijan, 2024-07-16:
@BeataZdunczyk Should be rather easy task. The kwd from dts-lib should be replaced from the one in lib/flash.robot for internal flashing.

wiktormowinski, 2024-07-19:
done with the switcheroo: https://github.com/Dasharo/open-source-firmware-validation/pull/395

Tests affected by the change:
ECR022.001 - we always skip that one 
ECR023.001 - as above
DTS005.001 - qemu skips it, am i doing something wrong or is this one also redundant??

macpijan, 2024-07-22:
QEMU does not support flashrom yet, for sure.

It may be true that this particular test case is mostly not used right now.

closed 2025-04-15 by BeataZdunczyk