Dasharo/open-source-firmware-validation #148: Detect USB-C Alt mode entry via USB Billboard device

mkopec, 2023-12-13:
### The problem you're addressing (if any)

USB-C tests are hard, since so many signals and alt modes are supported.

USB Billboard device class is a simple USB 1.1 device that is created on Alt mode entry faliure. Use that to make testing easier and more reliable.

### Describe the solution you'd like

Check the billboard device for any errors, if no errors are foumd then we probably are in the right alt mode, using the right cable, in the right port

### Where is the value to a user, and who might that user be?

Testing is more reliable

### Additional context

https://community.infineon.com/t5/Knowledge-Base-Articles/Billboard-device-USB-Type-C-Need-and-implementation/ta-p/355713
https://www.usb.org/document-library/billboard-device-class-spec-revision-122-and-adopters-agreement

I guess it depends on the vendor actually implementing the billboard class correctly

macpijan, 2024-07-16:
@BeataZdunczyk I cannot comment on this much, need @mkopec to provide more context and next steps here.

mkopec, 2024-07-17:
I wanted to make use of USB billboard device for detecting usb-c errors, but the docks we test don't implement it correctly (they always return the same status, regardless if everything is working correctly or not). Closing as it doesn't really make sense to pursue this.

closed 2024-07-17 by mkopec