Dasharo/open-source-firmware-validation #154: Get Secure Boot Menu Construction returns wrong menu

m-iwanicki, 2023-12-15:
### Device

QEMU

### RTE version

QEMU

### Brief summary

Get Secure Boot Menu Construction returns Advanced Secure Boot Keys Management menu instead of Secure Boot Configuration

### How reproducible

Every time

### How to reproduce

* Use main branch
* Initialize environment
* Create qemu-data dir `mkdir qemu-data`
* Create disk `qemu-img create -f qcow2 qemu-data/hdd.qcow2 20G`
* Create empty ubuntu.iso file `touch qemu-data/ubuntu.iso`
* run `./scripts/ci/qemu-run.sh graphic os` or `./scripts/ci/qemu-run.sh graphic os_install`
* run test `robot -L TRACE -v rte_ip:127.0.0.1 -v snipeit:no -v config:qemu dasharo-security/secure-boot.robot`

Bug also happens when using normal ubuntu.iso (for example with Ubuntu 22.04.03) installed or not.

### Expected behavior

After resetting secure boot keys to default and returning to previous menu Get Secure Boot Menu Construction should return
```${sb_menu} = ['Current Secure Boot State Disabled', 'Enable Secure Boot [ ]', 'Secure Boot Mode <Standard Mode>']```

### Actual behavior

Should return Secure Boot Configuration menu
```${sb_menu} = ['> Reset to default Secure Boot Keys', '> Erase all Secure Boot Keys', 'Individual key management:', '> PK Options', '> KEK Options', '> DB Options', '> DBX Options', 'v']```

### Link to screenshots or logs

[image]

### Additional context

Doesn't happen when using `./scripts/ci/qemu-run.sh graphic firmware`

### Solutions you've tried

Fix: Adding `Sleep    3s` in `lib/secure-boot-lib.robot` file in `Make Sure That Keys Are Provisioned` keyword between `Reset To Default Secure Boot Keys    ${advanced_menu}` and `Exit From Current Menu`.
It takes about 2-3 seconds to reset keys to default after choosing yes in this step:
[image]

m-iwanicki, 2024-01-29:
Adding `Read From Terminal Until    Esc=Exit` to the end of `Reset To Default Secure Boot Keys` should fix this issue ([754c180](https://github.com/Dasharo/open-source-firmware-validation/pull/194/commits/754c180a820e4e8826bd2d004c96000e5c256829)), previous solution only worked in that one keyword.

macpijan, 2024-07-16:
@BeataZdunczyk This can be grouped together with these big secure boot PRs.