Dasharo/open-source-firmware-validation #156: Tests dasharo-security/tpm2-commands.robot must not assume both PCR banks are always active

miczyg1, 2023-12-19:
### Device

Any

### Dasharo version

Any

### Affected component(s) or functionality

Test cases in dasharo-security/tpm2-commands.robot in OSFV.

### Brief summary

Test cases in dasharo-security/tpm2-commands.robot in OSFV test for SHA1 and SHA256 PCRs presence assuming both are always enabled. The test case should take into account platforms that have only fTPM. fTPM may have only one bank active on Intel platforms, so this test has 0% chance to pass.

### How reproducible

Always on platforms with PTT.

### How to reproduce

Run test suite on MSI platforms or Protectli V1x10

### Expected behavior

Test checks for active banks and verified the PCRs based on active banks instead.

`TPMCMD001.001 Check if both SHA1 and SHA256 PCRs are enabled` is only valid for discrete TPMs which typically have both SHA1 and SHA256 banks active simultaneously.

### Actual behavior

Test assume that SHA1 and SHA256 banks are always active which leads to a test failure on platforms with fTPM/

### Additional context

```

Checking if tpm2-tools is installed...

Package tpm2-tools is installed
TPMCMD001.001 Check if both SHA1 and SHA256 PCRs are enabled (Ubun... | FAIL |
'selected-pcrs:
  - sha1: [ ]
  - sha256: [ 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23 ]
' does not contain 'sha1: [ 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23 ]'
------------------------------------------------------------------------------
TPMCMD002.001 PCRREAD Function Verification (Ubuntu 22.04) :: This... | PASS |
------------------------------------------------------------------------------
TPMCMD003.001 PCREXTEND And PCRRESET Functions (Ubuntu 22.04) :: T... | FAIL |
'  sha1:
  sha256:
    0 : 0xDCC79DDA7F2549424CFB20AE2EF68E5A391144E26D2E90797B50C7BF81DB6A06
    1 : 0x5B79869F1908B6E52245E9A04204BC22CC7F75E965B2DCC2DD55E23D510A0192
    2 : 0x09730E5EDB26B3BEC7467D52AA671930ED888A9549594381F2E68025C2122FC4
    3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    4 : 0x7BCE664F8B2E0124B892A883EE3D7634FB7F5790A39D92C7B778F3CEEDE74FCF
    5 : 0x3B8729EF3E40DCB18F1809CB89C37AF7631CA4A1A22530AF7E2290415648343F
    6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    7 : 0xF550489E1E344C3BA21B71D4683A2A08DA95E9898B8D651456312460A5B8DC79
    8 : 0x77F3AD8574002524A137694E688716289F98FC02D78E041B2E6BA112E42E197B
    9 : 0x9A3F9EA0FFA4F62E1CB047BB1E1A2C0AA13691263E00DF04BC14E55F76635E3C
    10: 0x01D9ABC1485E7325E22E72D9FD478D6D7F7F07B5B652A70BF2CD7D44A1D68203
    11: 0x0000000000000000000000000000000000000000000000000000000000000000
    12: 0x0000000000000000000000000000000000000000000000000000000000000000
    13: 0x0000000000000000000000000000000000000000000000000000000000000000
    14: 0x306F9D8B94F17D93DC6E7CF8F5C79D652EB4C6C4D13DE2DDDC24AF416E13ECAF
    15: 0x0000000000000000000000000000000000000000000000000000000000000000
    16: 0x0000000000000000000000000000000000000000000000000000000000000000
    17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    23: 0x44F12027AB81DFB6E096018F5A9F19645F988D45529CDED3427159DC0032D921
' does not contain '23: 0x3D96EFE6E4A9ECB1270DF4D80DEDD5062B831B5A'
------------------------------------------------------------------------------
TPMCMD003.002 PCREXTEND And PCRRESET Functions - locality protecti... | PASS |
------------------------------------------------------------------------------
TPMCMD004.001 PCREVENT Function (Ubuntu 22.04) :: This test aims t... | FAIL |
'  sha1:
  sha256:
    0 : 0xDCC79DDA7F2549424CFB20AE2EF68E5A391144E26D2E90797B50C7BF81DB6A06
    1 : 0x5B79869F1908B6E52245E9A04204BC22CC7F75E965B2DCC2DD55E23D510A0192
    2 : 0x09730E5EDB26B3BEC7467D52AA671930ED888A9549594381F2E68025C2122FC4
    3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    4 : 0x7BCE664F8B2E0124B892A883EE3D7634FB7F5790A39D92C7B778F3CEEDE74FCF
    5 : 0x3B8729EF3E40DCB18F1809CB89C37AF7631CA4A1A22530AF7E2290415648343F
    6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    7 : 0xF550489E1E344C3BA21B71D4683A2A08DA95E9898B8D651456312460A5B8DC79
    8 : 0x77F3AD8574002524A137694E688716289F98FC02D78E041B2E6BA112E42E197B
    9 : 0x9A3F9EA0FFA4F62E1CB047BB1E1A2C0AA13691263E00DF04BC14E55F76635E3C
    10: 0x01D9ABC1485E7325E22E72D9FD478D6D7F7F07B5B652A70BF2CD7D44A1D68203
    11: 0x0000000000000000000000000000000000000000000000000000000000000000
    12: 0x0000000000000000000000000000000000000000000000000000000000000000
    13: 0x0000000000000000000000000000000000000000000000000000000000000000
    14: 0x306F9D8B94F17D93DC6E7CF8F5C79D652EB4C6C4D13DE2DDDC24AF416E13ECAF
    15: 0x0000000000000000000000000000000000000000000000000000000000000000
    16: 0x0000000000000000000000000000000000000000000000000000000000000000
    17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    23: 0x44F12027AB81DFB6E096018F5A9F19645F988D45529CDED3427159DC0032D921
' does not contain '23: 0x3D96EFE6E4A9ECB1270DF4D80DEDD5062B831B5A'

```

m-iwanicki, 2024-06-21:
@miczyg1 https://github.com/Dasharo/open-source-firmware-validation/pull/322 fixes this issue. Tests pass when only 1 bank is active. First test in this case is skipped.

In case neither SHA1 or SHA256 PCR bank is enabled then first test fails and tests 2-4 are skipped (not sure if needed).

Tested on Protectli v1210:

```
Checking if tpm2-tools is installed...

Package tpm2-tools is installed
TPMCMD001.001 Check if both SHA1 and SHA256 PCRs are enabled (Ubun... | SKIP |
Only one bank is enabled
------------------------------------------------------------------------------
TPMCMD002.001 PCRREAD Function Verification (Ubuntu 22.04) :: This... | PASS |
------------------------------------------------------------------------------
TPMCMD003.001 PCREXTEND And PCRRESET Functions (Ubuntu 22.04) :: T... | PASS |
------------------------------------------------------------------------------
TPMCMD003.002 PCREXTEND And PCRRESET Functions - locality protecti... | PASS |
------------------------------------------------------------------------------
TPMCMD004.001 PCREVENT Function (Ubuntu 22.04) :: This test aims t... | PASS |
------------------------------------------------------------------------------
TPMCMD005.001 CREATEPRIMARY Function Verification (Ubuntu 22.04) :... | PASS |
------------------------------------------------------------------------------
TPMCMD006.001 NVDEFINE and NVUNDEFINE Functions Verification (Ubun... | PASS |
------------------------------------------------------------------------------
TPMCMD007.001 CREATE Function (Ubuntu 22.04) :: This test aims to ... | PASS |
------------------------------------------------------------------------------
TPMCMD007.002 CREATELOADED Function (Ubuntu 22.04) :: This test ai... | PASS |
------------------------------------------------------------------------------
TPMCMD008.001 Signing the file (Ubuntu 22.04) :: Check whether the... | PASS |
------------------------------------------------------------------------------
TPMCMD009.001 Encryption and Decryption of the file (Ubuntu 22.04)... | SKIP |
TPM doesn't supports TPM2_EncryptDecrypt nor TPM2_EncryptDecrypt2
------------------------------------------------------------------------------
TPMCMD010.001 Hashing the file (Ubuntu 22.04) :: Check whether the... | PASS |
------------------------------------------------------------------------------
TPMCMD011.001 Performing HMAC operation on the file (Ubuntu 22.04)... | PASS |
------------------------------------------------------------------------------
Tpm2-Commands                                                         | PASS |
13 tests, 11 passed, 0 failed, 2 skipped
==============================================================================
```

Test on QEMU with only SHA512 bank enabled (and temporarily removed `tpm2_pcrallocate` from Suite Setup)

```
TPMCMD001.001 Check if both SHA1 and SHA256 PCRs are enabled (Ubun... | FAIL |
'False' should be true.
------------------------------------------------------------------------------
TPMCMD002.001 PCRREAD Function Verification (Ubuntu 22.04) :: This... | SKIP |
No PCR banks enabled
------------------------------------------------------------------------------
TPMCMD003.001 PCREXTEND And PCRRESET Functions (Ubuntu 22.04) :: T... | SKIP |
No PCR banks enabled
------------------------------------------------------------------------------
TPMCMD003.002 PCREXTEND And PCRRESET Functions - locality protecti... | SKIP |
No PCR banks enabled
------------------------------------------------------------------------------
TPMCMD004.001 PCREVENT Function (Ubuntu 22.04) :: This test aims t... | SKIP |
No PCR banks enabled
------------------------------------------------------------------------------
TPMCMD005.001 CREATEPRIMARY Function Verification (Ubuntu 22.04) :... | PASS |
------------------------------------------------------------------------------
(...)
```

m-iwanicki, 2024-06-21:
Or would it better to change first test from `Check if both SHA1 and SHA256 PCRs are enabled` to `Check if either SHA1 or SHA256 PCRs are enabled`?

macpijan, 2024-06-21:
It makes sense to me

closed 2024-06-21 by macpijan