Dasharo/open-source-firmware-validation #159: Factor power control, input and output methods into separate libraries

mkopec, 2023-12-19:
### The problem you're addressing (if any)

Right now it's a bit messy with each DUT using a different set of control methods for power, serial, wake etc, which are selected using flags (DUT_CONNECTION_METHOD etc)

### Describe the solution you'd like

Factor out each implementation into a separate library with a defined interface

i.e. `wakeonlan` and `rte` libraries for waking up, `rte` and `pikvm` for terminal output, `rte` and `sonoff` for power control, etc.

### Where is the value to a user, and who might that user be?

Makes tests simpler by having a set of well known kwds for things that are platform-specific

### Describe alternatives you've considered

do nothing?

macpijan, 2024-07-16:
@BeataZdunczyk  This may need a bit more work to evaluate all of the places needing to be changed, but probably still valid. We need to build more concrete plan here.

A good example of how to approach this is how the options lib has been implemented: https://github.com/Dasharo/open-source-firmware-validation/tree/develop/lib/options

philipandag, 2024-11-18:
The Power On / Power Cycle On and similar keywords behave differently on many devices.
For example Running `Power On` or `Power Cycle On` on a device without a power control, (laptops) does nothing.

On other devices running such keyword causes the device to turn off and boot with the firmware read from the flash memory. The tests which flash the devices depend on it and it might be surprising that running a Power On keyword won't cause the newly flashed firmware to be booted.

philipandag, 2025-01-20:
> For example Running Power On or Power Cycle On on a device without a power control, (laptops) does nothing.

The fact that Power On works differently on some platfoms is wrong. The keyword was documented in https://github.com/Dasharo/open-source-firmware-validation/pull/664 because of https://github.com/Dasharo/open-source-firmware-validation/issues/583. If the implementation is changed to work the same on every platform, the documentation should be updated