Dasharo/open-source-firmware-validation #162: Reset to defaults tests should be customizable

macpijan, 2023-12-20:
In test title, we should not say which is expected default, as it may be platform dependent. 

These should be configurable in platform configs.

It also applies to numeric values as well. E.g. currently tests expects timeout of 500, while in the v1.1.3 MSI release we set 600.

macpijan, 2024-07-16:
@BeataZdunczyk 

- [ ] Confirm what do we set in MSI configs in coreboot (is it 600?)
- [ ] Do we set this in any other platforms?
- [ ] Change the test name to avoid exact value in the title
- [ ] Add expected default value to platform configs

wiktormowinski, 2024-07-19:
Ok to clear things up, my job here is to change the current values at the end of each test to a correct config value, and if some platform config doesn't have such information I need to add it myself.
 

If so I need to know what reference material should I follow while filling up the configs with defaults?

macpijan, 2024-11-25:
Some effort done in: https://github.com/Dasharo/open-source-firmware-validation/pull/391

Maybe we will come back here again. It is not the most important issue right now.