Dasharo/open-source-firmware-validation #165: Add second sonoff to laptop testing station

mkopec, 2023-12-20:
### The problem you're addressing (if any)

The docking station tests require power control over the lapotp and dock. Right now we have to manually move the sonoff to the docking station PSU so that the laptop can be controlled via power on AC

### Describe the solution you'd like

Second sonoff for simulating docking hotplug events.

### Where is the value to a user, and who might that user be?

Better tests with less manual intervention.

macpijan, 2024-07-16:
@BeataZdunczyk Adding `test-stand` label, but this is probably very low priority as we do not use this stand right now.

BeataZdunczyk, 2024-07-19:
> probably very low priority as we do not use this stand right now.

Yes, it is not applicable right now.