Dasharo/open-source-firmware-validation #166: Laptop tests should be done with battery connected

mkopec, 2023-12-20:
### The problem you're addressing (if any)

Currently we have to test with the battery disconnected, to have power control via sonoff. This is less than ideal, especially since a tiny % of users will be usint their laptop without a battery connected, and we can't test features like HPB or battery / charging status without the battery connected.

### Describe the solution you'd like

OOB power button control. I am experimenting with USB-PD based solutions using https://github.com/mkopec/tipd (very WIP), the real solution will probably use a Twonkie.

### Where is the value to a user, and who might that user be?

More automation, less manual steps, more things can be tested, tests will be closer to the actual real world use cases

### Describe alternatives you've considered

WOL is good as a solution for wakeup from Sx but doesn't let us reset a platform completely when it locks up.

philipandag, 2024-07-22:
I have heard from @mkopec that the issue no longer applies because we are now testing novacustom laptops via ssh and testing with battery connected is now possible.  @mkopec please confirm if that is true and close the issue if so :)

closed 2024-07-22 by philipandag

mkopec, 2024-07-22:
Indeed, this issue is no longer applicable.