Dasharo/open-source-firmware-validation #173: Memory-Profile test: AttributeError

filipleple, 2023-12-22:
### Device

MSI zx90

### RTE version

e1066d329f64f2e8daf2a438c7cd92452c18fa04

### Affected component(s) or functionality

Memory-Profile test 

### Brief summary

Memory-Profile test can't switch back to JEDEC; fails with `AttributeError: 'bool' object has no attribute 'startswith'`

### How reproducible

100%

### How to reproduce

Run the Memory-Profile test

### Expected behavior

Test should pass

### Actual behavior

Test fails

### Link to screenshots or logs

[Full HTML log](https://cloud.3mdeb.com/index.php/s/nK9DfsiWqwcL7X2)

mkopec, 2024-01-17:
The test returns either a Boolean or a String depending on the test result, hence the type error.

macpijan, 2024-07-16:
@BeataZdunczyk Might be still relevant, needs reproduction.

philipandag, 2024-08-13:
I can still reproduce the issue. Working on it.

philipandag, 2024-08-16:
https://github.com/Dasharo/open-source-firmware-validation/pull/458

BeataZdunczyk, 2024-08-21:
No reviewers assigned to this PR, @philipandag

SebastianCzapla, 2025-04-24:
Closing as the https://github.com/Dasharo/open-source-firmware-validation/pull/458 was merged.

closed 2025-04-24 by SebastianCzapla