Dasharo/open-source-firmware-validation #180: Performance tests fail on MSI zx90

filipleple, 2023-12-28:
### Device

MSI zx90

### RTE version

7ae61820d46700be2667dc5d24a81ac1146451b9

### Brief summary

Performance tests repeatedly fail due to Telnet errors/failure to catch tianocore string on zx90 boards

### How to reproduce

Run the Performance tests

### Expected behavior

They should pass or at least fail due to firmware bugs

### Actual behavior

They fail due to telnet errors

### Link to screenshots or logs

[logs](https://cloud.3mdeb.com/index.php/s/w7j4W26P5ib2pEc) for different platforms

macpijan, 2024-07-16:
@BeataZdunczyk We might check if this is still relevant after we finalize this branch with improvements from current testing: https://github.com/Dasharo/open-source-firmware-validation/pull/361

philipandag, 2024-08-14:
This issue definitely still exists on develop and not only during performance tests. It seems to be a firmware/hardware error because I manged to run tests on msiz690-ddr5 but every other msi board fails because of this issue.

philipandag, 2024-08-14:
Turns out I was just leaving the platforms in a bricked state after setting XMP profiles which are not supported but still available in the setup menu causing them to become bricked. I suspect that is the case here too. Too bad we don't have any log which does not fail on the first try to read anything from serial. I will see if this issue still appears when the platforms are not bricked.

philipandag, 2024-08-16:
https://github.com/Dasharo/open-source-firmware-validation/pull/461