Dasharo/open-source-firmware-validation #190: USB device detection tests use unstable device names

mkopec, 2024-01-09:
### Device

Any

### RTE version

n/a

### Affected component(s) or functionality

USB-C docking station tests (and probably also other USB tests)

### Brief summary

Tests look for a specific device name in `lsusb` like `Genesys Logic, Inc. USB3.2 Hub`. These names come from the OS's USB ID database and are subject to change.

### How reproducible

100%

### How to reproduce

Run keyword `Detect Docking Station USB Devices In Linux` on Wavlink UMD05 Pro Rev.e on newer Ubuntu versions

### Expected behavior

Test detects devices correctly

### Actual behavior

Test fails because one device changed name and it's not recognized

### Link to screenshots or logs

```
UTC008.001 Docking station detection after coldboot (Ubuntu 22.04)... .Please make sure the docking station connected is WL-UMD05 Pro Rev.E and press ENTER

Manual step confirmed
UTC008.001 Docking station detection after coldboot (Ubuntu 22.04)... | FAIL |
'Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 04f2:b6fb Chicony Electronics Co., Ltd Chicony USB2.0 Camera
Bus 003 Device 007: ID 2109:4817 VIA Labs, Inc. USB2.0 Hub
Bus 003 Device 009: ID 0c76:1707 JMTek, LLC. USB Audio
Bus 003 Device 008: ID 1d5c:7102 Fresco Logic Generic Billboard Device
Bus 003 Device 005: ID 067b:2586 Prolific Technology, Inc. USB 2.0 Hub
Bus 003 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 006: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 002: ID 1d6b:0104 Linux Foundation Multifunction Composite Gadget
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0bda:0306 Realtek Semiconductor Corp. USB3.0 Card Reader
Bus 002 Device 006: ID 13fe:6700 Kingston Technology Company Inc. USB DISK 3.0
Bus 002 Device 005: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 003: ID 2109:1817 VIA Labs, Inc. USB3.0 Hub
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub' does not contain 'Genesys Logic, Inc. USB3.2 Hub'
```

`Genesys Logic, Inc. GL3523 Hub` = `Genesys Logic, Inc. USB3.2 Hub`.

### Additional context

Simply use the PID:VID combination that doesn't change with OS updates.

mkopec, 2024-01-10:
Another one:

```
Bus 003 Device 004: ID 5986:9102 Acer, Inc BisonCam,NB Pro [...] does not contain 'Camera'
```

filipleple, 2024-06-07:
Issue resolved as of [this](https://github.com/Dasharo/open-source-firmware-validation/pull/311/commits/9ed12c4363426b5a1a03252a8287f40da76db20c) commit.

Log available [here](https://cloud.3mdeb.com/index.php/s/faEY74XboZSdeGA).

closed 2024-06-07 by filipleple