Dasharo/open-source-firmware-validation #195: Tests fail due to missing packages

mkopec, 2024-01-16:
### Device

NovaCustom NS70MU

### RTE version

n/a

### Affected component(s) or functionality

Wi-Fi / BT tests

### Brief summary

Wi-Fi / BT tests fail due to missing `lspci` package

### How reproducible

100%

### How to reproduce

- install ubuntu 22.04
- start running tests on a clean installation

### Expected behavior

tests install all missing utils automatically

### Actual behavior

tests fail because `lspci` is missing

### Link to screenshots or logs

[image]

### Solutions you've tried

- don't depend on lspci?
- take care of installing all utilities in suite setup. `ethtool` is also not installed OOTB, so warmboot tests using WoL fail.

mkopec, 2024-01-16:
another one: `fwts` in `SUSP005.002`

macpijan, 2024-05-10:
@artur-rs this would be trivial fix

JanPrusinowski, 2024-05-27:
PR dostepny: https://github.com/Dasharo/open-source-firmware-validation/pull/297

closed 2024-08-14 by macpijan