Dasharo/open-source-firmware-validation #196: DTS tests fail on laptops

mkopec, 2024-01-16:
### Device

NovaCustom NS70MU

### RTE version

n/a

### Affected component(s) or functionality

Tests using DTS (for example `LCM001.001`)

### Brief summary

Tests expect DTS console to appear on serial, but that's not possible on laptops due to how the EC console works.

### How reproducible

100%

### How to reproduce

- run `LCM001.001` on NS70MU

### Expected behavior

test work

### Actual behavior

tests fail due to timeout, right after loading DTS via iPXE

### Link to screenshots or logs

[image]

### Additional context

I guess we could try to blindly enable the SSH server via PiKVM and then have tests use SSH in DTS instead of telnet, or figure out a better laptop serial console connection

macpijan, 2024-05-10:
We do this, and it used to work on MSIs at least.
How it was done: https://github.com/Dasharo/open-source-firmware-validation/issues/282#issuecomment-2104517338

m-iwanicki, 2024-05-10:
If I understand [this](https://github.com/Dasharo/open-source-firmware-validation/blob/787d7a14c73ecda7920dfd3efa310062174d9aec/lib/dts-lib.robot#L25) comment correctly it means that on serial we only have [keyboard] input and no [console] output?

It's that's true then after this comment there should be no reading from terminal until we connect via ssh but next keyword waits for menu to be seen.

```
# Wait for the menu to be loaded on serial
        Read From Terminal Until    Enter an option:
```

m-iwanicki, 2024-06-21:
Should be fixed by https://github.com/Dasharo/open-source-firmware-validation/pull/317.
Tested on msi (`DTS003.001`) to check if my changes didn't mess anything up.
Didn't test on laptop as the one in the lab is not available.

SebastianCzapla, 2025-04-30:
With a laptop coming back to lab setup recently, it's possible to retest this issue now