Dasharo/open-source-firmware-validation #198: Allow for running tests without RTE

macpijan, 2024-01-16:
### The problem you're addressing (if any)

Currently, it is assumed in the environment, that each setup needs RTE. This is not true anymore. In some cases we already have some workarounds for that (e.g. in QEMU).

### Describe the solution you'd like

Some tests can be run with no power control at all (we power up the device manually, the tests are executed over serial or SSH).

Some tests can be run with manual power control.

### Where is the value to a user, and who might that user be?

Existing users of OSFV can get rid of workarounds.

New users of OSFV can actually benefit from it more, by running tests themselves. At this moment, it is very demanding on the hardware side, as very few users have RTE, or other equipment.

### Additional context

Some more context can be extracted from this discussion: https://github.com/Dasharo/dasharo-issues/issues/468

macpijan, 2024-01-16:
Another use-case we faced was testing the [TwPM](https://twpm.dasharo.com/). We did not need RTE here, using it made our setup more complex and testing more time-consuming. We have applied the following diff to disable the power control via RTE, it was assumed the the DUT was powered on with SSH enabled, which can be the case for many end users as well:

```
diff --git a/dasharo-security/tpm2-commands.robot b/dasharo-security/tpm2-commands.robot
index 031f1d4a533b..aec1c174b82e 100644
--- a/dasharo-security/tpm2-commands.robot
+++ b/dasharo-security/tpm2-commands.robot
@@ -248,8 +248,8 @@ TPM2 Test Setup
 
 TPM2 Suite Setup
     Prepare Test Suite
-    Power On
-    Boot System Or From Connected Disk    ubuntu
+    #Power On
+    #Boot System Or From Connected Disk    ubuntu
     Login To Linux
     Switch To Root User
     Detect Or Install Package    tpm2-tools
diff --git a/os-config/ubuntu-credentials.robot b/os-config/ubuntu-credentials.robot
index def24c2674fb..4755a9896f2e 100644
--- a/os-config/ubuntu-credentials.robot
+++ b/os-config/ubuntu-credentials.robot
@@ -1,5 +1,5 @@
 *** Variables ***
-${UBUNTU_USERNAME}=         user
+${UBUNTU_USERNAME}=         ubuntu
 ${UBUNTU_PASSWORD}=         ubuntu
-${UBUNTU_USER_PROMPT}=      ubuntu@3mdeb:~$
-${UBUNTU_ROOT_PROMPT}=      root@3mdeb:/home/ubuntu#
+${UBUNTU_USER_PROMPT}=      ubuntu@ubuntu-VP4670:~$
+${UBUNTU_ROOT_PROMPT}=      root@ubuntu-VP4670:/home/ubuntu#
diff --git a/platform-configs/protectli-vp4670.robot b/platform-configs/protectli-vp4670.robot
index 8c33ecd4c1fa..e21cc8bbfb97 100644
--- a/platform-configs/protectli-vp4670.robot
+++ b/platform-configs/protectli-vp4670.robot
@@ -3,7 +3,7 @@ Resource    ../os-config/ubuntu-credentials.robot
 
 
 *** Variables ***
-${DUT_CONNECTION_METHOD}=                           Telnet
+${DUT_CONNECTION_METHOD}=                           SSH
 ${PAYLOAD}=                                         tianocore
 ${RTE_S2_N_PORT}=                                   13541
 ${FLASH_SIZE}=                                      ${16*1024*1024}
 ```

pietrushnic, 2024-01-16:
@macpijan I would love to see this as an option in executing OSFV. This was my request from the beginning of the project. It would be great if this issue would be assigned to some publicly known milestone with deadline.

WiktorG351, 2024-04-25:
I'm thinking a command-line passed variable could be implemented, that allows for running a test without RTE (for example: `-v RTE:NONE`).

Some keywords that would definitely need to be changed are:

* `Power Cycle On`
* `Turn On Power Supply`
* etc.

In these something like this would have to be added:

```
IF  '${RTE}' == 'NONE'
    Run     echo -n "Turn power ON/OFF. Press ENTER once done in order to continue... " && read ignore
    RETURN
END
```

The serial connection seems to be a lot simpler, it appears that only 1 keyword is relevant here, that being `Serial Setup`.

All we would have to do is provide the end user a guide on how to connect directly from host to the serial port of the machine they are trying to run the tests on, and how to use `ser2net` in such a way so that the existing test infrastructure can be reused.

artur-rs, 2024-04-25:
@WiktorG351 

1. I'm thinking about the usage of already existing flags (we should try to not introduce more variables whenever we can). What about setting `rte_ip` to `NONE` - is it doable? The second idea is to use `${POWER_CTRL}` (that can be found in the `platform-configs/` files) and set it to `manual` or something similar.
1. It's better to use `Dialogs` standard RF lib to introduce interaction with the user: https://robotframework.org/robotframework/latest/libraries/Dialogs.html

macpijan, 2024-07-16:
@BeataZdunczyk This should rather be a separate power control library (`manual` power control), to be integrated while following approach listed in this issue: https://github.com/Dasharo/open-source-firmware-validation/issues/159

macpijan, 2024-07-16:
>     It's better to use Dialogs standard RF lib to introduce interaction with the user: https://robotframework.org/robotframework/latest/libraries/Dialogs.html

+1 for Dialogs

krystian-hebel, 2024-09-26:
Loosely related: issue templates ask for RTE branch and commit in first question, which usually isn't relevant. I believe this may be a leftover from before the repo was renamed.