Dasharo/open-source-firmware-validation #201: Robotidy fails with exception all the time.

m-iwanicki, 2024-01-17:
### Device

N/A

### RTE version

N/A

### Affected component(s) or functionality

pre-commit hooks

### Brief summary

robotidy fails with exception when trying to commit

### How reproducible

Every time 

### How to reproduce

Unknown, happened to me when trying to cherry-pick commits from [non-rootfs-encryption-tpm](https://github.com/Dasharo/open-source-firmware-validation/tree/non-rootfs-encryption-tpm) into [ami-encrypted-rootfs-tests](https://github.com/Dasharo/open-source-firmware-validation/tree/ami-encrypted-rootfs-tests).
After I resolved conflict in first commit I couldn't continue cherry-pick because of this error. After that it happened every time, on every branch.
Can't try to reproduce because after it happens first time then it happens even after deleting and recloning repo.

### Expected behavior

robotidy should fail or pass normally, not raise exceptions.

### Actual behavior

robotidy/python raised "TypeError: can only concatenate list (not "tuple") to list"

### Link to screenshots or logs

```
robotidy.................................................................Failed
- hook id: robotidy
- exit code: 1

Traceback (most recent call last):
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/bin/robotidy", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/rich_click/rich_command.py", line 126, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/decorators.py", line 30, in wrapper
    raise err
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/decorators.py", line 21, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/cli.py", line 480, in cli
    global_config = config_module.MainConfig(cli_config)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/config.py", line 228, in __init__
    self.default_loaded = Config.from_raw_config(self.default)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/config.py", line 381, in from_raw_config
    return cls(
           ^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/config.py", line 323, in __init__
    self.load_transformers(transformers_config, force_order, target_version, skip)
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/config.py", line 402, in load_transformers
    transformers = load_transformers(
                   ^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/transformers/__init__.py", line 480, in load_transformers
    for container in import_transformer(name, transformers_config, skip):
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/transformers/__init__.py", line 324, in import_transformer
    yield create_transformer_instance(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/transformers/__init__.py", line 346, in create_transformer_instance
    positional, named, argument_names = resolve_args(short_name, spec, args, skip, handles_skip=handles_skip)
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/transformers/__init__.py", line 424, in resolve_args
    argument_names = resolve_argument_names(spec.argument_names, handles_skip)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miwanicki/.cache/pre-commit/repoo9k6miyd/py_env-python3/lib/python3.11/site-packages/robotidy/transformers/__init__.py", line 368, in resolve_argument_names
    return new_args + argument_names
           ~~~~~~~~~^~~~~~~~~~~~~~~~
TypeError: can only concatenate list (not "tuple") to list
Fatal exception occurred. You can create an issue at https://github.com/MarketSquare/robotframework-tidy/issues . Thanks!
```

### Additional context

After error happens first time then nothing helps not even deleting whole repository
Identical problem happened to @DaniilKl and @filipleple on different branches.
argument_names value when error happens is `('order', 'create_comment_section')`

### Solutions you've tried

Working:
* change `return new_args + argument_names` to `return [*new_args, *argument_names]` in ~/.cache/pre-commit/.../robotidy/transformers/__init__.py

Not working:
* deleted ~/.cache along with osfv repo, cloned and initialized repo and venv again. Error still happened on every commit, even on main branch

mkopec, 2024-01-17:
Maybe you need to run the git command while inside the robot venv?

```
source venv/bin/activate
```

DaniilKl, 2024-01-17:
> Maybe you need to run the git command while inside the robot venv?
> 
> ```
> source venv/bin/activate
> ```

I was having the same error. Unfortunately did not notice when it arose. I am shure I had `venv` launched.

m-iwanicki, 2024-01-17:
@mkopec I was in venv every time. I cloned repo couple times and went through "Initializing environment" step every time.

filipleple, 2024-01-19:
Also happens for me, running in the `robot-venv`, trying to commit on [this branch](https://github.com/Dasharo/open-source-firmware-validation/pull/139)

closed 2024-01-24 by macpijan