Dasharo/open-source-firmware-validation #206: "USB002.001 USB keyboard detected in FW" gives false positive result

Psotas, 2024-02-06:
### Device

V1210

### RTE version

Dasharo v0.9.2

### Brief summary

`USB002.001 USB keyboard detected in FW` gives false positive result

### How to reproduce

Run test `USB002.001 USB keyboard detected in FW`

### Expected behavior

Detect keyboard in FW

### Actual behavior

Enter to `Device Manager` menu.

### Link to screenshots or logs

```
USB002.001 USB keyboard detected in FW
    [Documentation]    Check whether the external USB keyboard is detected
    ...    correctly by the firmware and all basic keys work
    ...    according to their labels.
    Power On
    ${setup_menu}=    Enter Setup Menu Tianocore And Return Construction
    ${device_mgr_menu}=    Enter Submenu From Snapshot And Return Construction
    ...    ${setup_menu}
    ...    Device Manager
    Save Changes And Reset
```

macpijan, 2024-02-06:
true, while it makes some sense on devices with PiKVM (where USB kbd for input is used), it does not make much sense for serial-controlled devices

BeataZdunczyk, 2024-07-16:
@PLangowski what is the status here? Linked PR is closed. Is there a fix? We should link PR with the fix, and close it then. @artur-rs cc

PLangowski, 2024-07-16:
@BeataZdunczyk The PR got closed automatically and couldn't be reopened. I created a new one https://github.com/Dasharo/open-source-firmware-validation/pull/383

closed 2024-07-30 by m-iwanicki