Dasharo/open-source-firmware-validation #207: CPU frequency/temperature tests fail on data parsing for v1210

Psotas, 2024-02-08:
### Device

V1210

### RTE version

Dasharo v0.9.2

### Brief summary

Improve CPU freq/temp tests to work properly for v1210

### How to reproduce

Run `cpu-frequency.robot` and `cpu-temperature.robot` tests.

### Expected behavior

Test pass

### Actual behavior

Tests fail on data parsing

### Link to screenshots or logs

[CPU-freq-temp_log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/14208133/CPU-freq-temp_log.zip)

CPF002.001, CPF002.001 and CPF004.001:
```
Traceback (most recent call last):
  None
RuntimeError: Evaluating expression 're.sub(r'(?s)[^0-9]*([1-9][0-9]*)[,.][0-9]+MHz', r'\1', $frequency)' failed: SyntaxError: invalid syntax (<string>, line 1)
```

CPT001.001 and CPT002.001:
```
Traceback (most recent call last):
  None
RuntimeError: '105.0°C)
' cannot be converted to a floating point number: ValueError: could not convert string to float: '105.0°C)\n'
```

macpijan, 2024-05-10:
@pkubaj please update the state here after retesting V1210 next week

macpijan, 2024-07-11:
This is still valid on revision: 04e410031775b1ee51d8b9c97915d27696856bb8

closed 2024-07-12 by macpijan