Dasharo/open-source-firmware-validation #209: Platform configs rework - 1st step

macpijan, 2024-03-02:
There is a massive duplication and unused content in the platform config. Also, more readable and modern format for variables could be used (e.g. YAML).

The goals of the 1st step:
- [ ] Drop unused variables
- [ ] Reduce duplication, by including variables hierarchically from multiple files, such as: default values, vendor values, platform values
- [ ] Create tools to ensure no unused variables are present in the future in the configuration files
- [ ] Plan next steps

macpijan, 2024-03-02:
Tool and pre-commit integration https://github.com/Dasharo/open-source-firmware-validation/pull/210/commits/ee23983021cb648fed8a2bee627110fabbd9ea06

Example output:

```
./scripts/check-unused-variables.sh platform-configs/msi-pro-z690-a-ddr5.robot
Found 61 defined variables not used in the test environment:
${ACCEPTED_%_NEAR_INITIAL_RPM}
${ALLOWED_BOOTING_FAILS}
${COREBOOT_BASE_PORT_SUPPORT}
${DASHARO_SERIAL_PORT_MENU_SUPPORT}
${DEBIAN_BOOTING}
${DEBIAN_BOOTING_ITERATIONS}
${DEVICE_BOOT_MEASURE_SUPPORT}
${DOCKING_STATION_UPLOAD_SUPPORT}
${DTS_FWUPD_FIRMWARE_UPDATE_SUPPORT}
${EDK2_IPXE_START_POS}
${FAN_CONTROL_MEASURE_INTERVAL}
${FAN_CONTROL_TEST_DURATION}
${FLASHING_BASIC_METHOD}
${FREEBSD_BOOTING}
${FREEBSD_BOOTING_ITERATIONS}
${INCORRECT_SIGNATURES_FIRMWARE}
${INITIAL_FAN_RPM}
${INSTALL_DEBIAN_USB_SUPPORT}
${INSTALL_UBUNTU_USB_SUPPORT}
${IPXE_STRING}
${NO_CHECK_SONOFF}
${OPNSENSE_SERIAL_BOOTING}
${OPNSENSE_SERIAL_BOOTING_ITERATIONS}
${OPNSENSE_VGA_BOOTING}
${OPNSENSE_VGA_BOOTING_ITERATIONS}
${PAYLOAD_STRING}
${PFSENSE_SERIAL_BOOTING}
${PFSENSE_SERIAL_BOOTING_ITERATIONS}
${PFSENSE_VGA_BOOTING}
${PFSENSE_VGA_BOOTING_ITERATIONS}
${PROXMOX_VE_BOOTING}
${PROXMOX_VE_BOOTING_ITERATIONS}
${SN_PATTERN}
${SOL_STRING}
${TEST_FAN_SPEED}
${TESTS_IN_FREEBSD_SUPPORT}
${TESTS_IN_OPNSENSE_SERIAL_SUPPORT}
${TESTS_IN_OPNSENSE_VGA_SUPPORT}
${TESTS_IN_PFSENSE_SERIAL_SUPPORT}
${TESTS_IN_PFSENSE_VGA_SUPPORT}
${TESTS_IN_PROXMOX_VE_SUPPORT}
${TESTS_IN_UBUNTU_SERVER_SUPPORT}
${THUNDERBOLT_CHARGING_SUPPORT}
${THUNDERBOLT_DOCKING_STATION_AUDIO_SUPPORT}
${THUNDERBOLT_DOCKING_STATION_DETECT_SUPPORT}
${THUNDERBOLT_DOCKING_STATION_DISPLAY_PORT}
${THUNDERBOLT_DOCKING_STATION_HDMI}
${THUNDERBOLT_DOCKING_STATION_KEYBOARD_SUPPORT}
${THUNDERBOLT_DOCKING_STATION_NET_INTERFACE}
${THUNDERBOLT_DOCKING_STATION_SUPPORT}
${THUNDERBOLT_DOCKING_STATION_UPLOAD_SUPPORT}
${THUNDERBOLT_DOCKING_STATION_USB_SUPPORT}
${UBUNTU_BOOTING}
${UBUNTU_BOOTING_ITERATIONS}
${UBUNTU_SERVER_BOOTING}
${UBUNTU_SERVER_BOOTING_ITERATIONS}
${USB_C_CHARGING_SUPPORT}
${USB_C_DISPLAY_SUPPORT}
${VBOOT_KEYS_GENERATING_SUPPORT}
${WINDOWS_BOOTING}
${WINDOWS_BOOTING_ITERATIONS}
```

macpijan, 2024-03-02:
Dropping unused variables: https://github.com/Dasharo/open-source-firmware-validation/pull/210/commits/ecc5192769a860eed0d6db3d8d5f1e025926c943

mkopec, 2024-04-08:
Working on reworking NCM configs in https://github.com/Dasharo/open-source-firmware-validation/pull/260

closed 2024-05-10 by macpijan