Dasharo/open-source-firmware-validation #215: Add `watchdog option` tests

Psotas, 2024-03-08:
### The problem you're addressing (if any)

No `watchdog option` tests

macpijan, 2024-03-10:
Related ticket: https://github.com/Dasharo/dasharo-issues/issues/734

PLangowski, 2024-05-07:
@macpijan could you provide a description of what should be done here?

artur-rs, 2024-05-07:
@PLangowski issue linked by Maciej shows the entire test case with steps to introduce the new test. First, check and propose the best place for the new test, then implement and verify it.

macpijan, 2024-05-08:
Sometimes it is a matter of checking what's there and some issues might be fixed already. For watchdog, it seems we already have test here: https://github.com/Dasharo/open-source-firmware-validation/blob/develop/dasharo-compatibility/watchdog.robot

PLangowski, 2024-05-08:
@macpijan We don't have a test that verifies that the watchdog resets the machine after a kernel crash.

PLangowski, 2024-05-08:
Unless it's the same case as in the second test

macpijan, 2024-05-08:
Right, got misled by the issue name. Adding a test after kernel crash would make very much sense. Reopening.

macpijan, 2024-05-08:
In terms of firmware, you should be able to get one from the CI in the linked PR.

PLangowski, 2024-05-08:
Yes, I've been searching for the artifacts from the [deployment](https://github.com/Dasharo/coreboot/actions/runs/8689984855/job/23828824675), but I can't find any