Dasharo/open-source-firmware-validation #218: `uefi-password.robot`  AttributeError: 'list' object has no attribute 'encode'

Psotas, 2024-03-08:
### Device

VP4650

### RTE version

Dasharo v1.2.0

### Brief summary

PSW002.001 Password setting mechanism correctness checking :: This... | FAIL | AttributeError: 'list' object has no attribute 'encode'

### How to reproduce

robot -L TRACE -v rte_ip:192.168.10.203 -v config:protectli-vp4650 -v device_ip:192.168.10.21 dasharo-security/uefi-password.robot 

### Expected behavior

PASS

### Actual behavior

AttributeError: 'list' object has no attribute 'encode'

### Link to screenshots or logs

[log-uefi-password.zip](https://github.com/Dasharo/open-source-firmware-validation/files/14538655/log-uefi-password.zip)

macpijan, 2024-03-10:
@Psotas Please check this commit: e144bcbf523f1c27231c15e0a4c68663e54970cc

Psotas, 2024-03-11:
@macpijan 
It is works.
I also improve teardown in commit.
[log-teardown-fixed.zip](https://github.com/Dasharo/open-source-firmware-validation/files/14555739/log-teardown-fixed.zip)

closed 2024-03-11 by macpijan