Dasharo/open-source-firmware-validation #225: Broken configs for novacustom-nv41* platforms

krystian-hebel, 2024-03-14:
### Device

novacustom-nv41*

### RTE version

does not apply

### Brief summary

All of novacustom-nv41* configs try to import file that doesn't exist

### How to reproduce

Start any test with any novacustom-nv41* config

### Expected behavior

Test suite succeeds

### Actual behavior

```
[ ERROR ] Error in file '/media/krystian/2171d05c-871e-4770-a24e-cc2073b9ab73/work/open-source-firmware-validation/platform-configs/novacustom-nv41mb.robot' on line 2: Resource file '../os/ubuntu_2204_credentials.robot' does not exist.
[ ERROR ] Error in file '/media/krystian/2171d05c-871e-4770-a24e-cc2073b9ab73/work/open-source-firmware-validation/platform-configs/novacustom-nv41mb.robot' on line 32: Setting variable '${DEVICE_UBUNTU_USERNAME}' failed: Variable '${UBUNTU_USERNAME}' not found.
```

### Link to screenshots or logs

[log_html_but_GH_doesnt_like_it.txt](https://github.com/Dasharo/open-source-firmware-validation/files/14599108/log_html_but_GH_doesnt_like_it.txt)

mkopec, 2024-04-08:
Working on it in https://github.com/Dasharo/open-source-firmware-validation/pull/260

macpijan, 2024-05-10:
@mkopec 

> Should probably document the test station stuff a bit

Do we have some documentation already?

What exactly do you have in mind? Can you summarize the documentation plan (if any) with checklist here?

macpijan, 2024-07-16:
@mkopec @filipleple I think we do already have some documentation for running on laptops, can you please link it here?
It should explain what is expected to work on laptops, and how to run these tests.

filipleple, 2024-07-17:
@macpijan [here](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/docs/novacustom.md)'s what we currently have on running tests on NVC laptops

philipandag, 2024-07-23:
@macpijan What else is there to do in this issue? The configs were fixed. Can it be closed?

mkopec, 2024-07-24:
We're running tests on laptops now, I do believe this issue can be closed

closed 2024-07-24 by mkopec