Dasharo/open-source-firmware-validation #234: CPF004.002 CPU with load runs on expected frequency (Windows 11) requires sshpass but doesn't install it

pkubaj, 2024-03-20:
### Device

VP4670 v2

### RTE version

1.1.0

### Brief summary

sshpass is not installed but required

### How reproducible

100%

### How to reproduce

```
robot -b command_log.txt -v snipeit:no -L TRACE -v rte_ip:192.168.10.14 -v device_ip:192.168.10.53 -v config:protectli-vp4670 -t "CPF004.002*" dasharo-performance/cpu-frequency.robot
```

### Expected behavior

Test should pass

### Actual behavior

```
CPF004.002 CPU with load runs on expected frequency (Windows 11) :... | FAIL |                 
'/bin/sh: 1: sshpass: not found' should be empty.                                              
```

### Link to screenshots or logs

[command_log.txt](https://github.com/Dasharo/open-source-firmware-validation/files/14666415/command_log.txt)

pkubaj, 2024-03-20:
Nevermind, it needed to be installed on the host that executed the test.

macpijan, 2024-07-12:
Integrated into: https://github.com/Dasharo/open-source-firmware-validation/pull/361

closed 2024-07-12 by macpijan