Dasharo/open-source-firmware-validation #250: VBO009.001 fails on apu

mkopec, 2024-03-28:
### Device

PC Engines apu4

### RTE version

v1.1.0

### Affected component(s) or functionality

Vboot test suite

### Brief summary

VBO009.001 fails on apu. Boots into the wrong OS and times out waiting for the Ubuntu shell prompt.

### How reproducible

100%

### How to reproduce

- Run Vboot test suite
- Check the results

### Expected behavior

- All tests pass

### Actual behavior

VBO009.001 fails

### Link to screenshots or logs

[log.zip](https://github.com/Dasharo/open-source-firmware-validation/files/14787781/log.zip)

macpijan, 2024-05-10:
@mkopec It passed manually? Do we know what was the problem?

macpijan, 2024-07-16:
@BeataZdunczyk @JanPrusinowski Another vboot-related issue. Added label to group them. Would be nice to extend the check/solve them together while we are here?

JanPrusinowski, 2024-10-07:
There were some issues with DUT.
Ubuntu was not installed.
After installing OS there were some issues with USB which caused unwanted logs to apear in the console.
DUT's IP must have changed as it turned out that it was different than in our config files.
There were issues with keywords which cloned repos from git.
There was an issue where `Make Sure That Flash Locks Are Disabled` kwd was missing... I have copied it from dcu.robot ad DUT does not use DCU.
Also there was a problem with flashing on the machine... Chip information was missing from the config.

All of the above changes/fixes tests could finally be run. However the original issue did not reproduce on the branch forked from the actual develop.

macpijan, 2024-10-23:
@JanPrusinowski This is not the proper approach here.

I have tracked it down a bit further in git history.

@filipleple Can you help me understand / fix these two commits:

1. https://github.com/Dasharo/open-source-firmware-validation/commit/39d65c703c86988be30f706f99051bd9f49a901b#diff-8d01f6edc2ae98aada8d51a7adb6dc18a7b43ab692f74e7dd01350cda86ce6a2R130
- you split the `Make Sure That Flash Locks Are Disabled` keywords into two options lib
- you keep the same documentation string for both
- but the functionality is entirely different in each of them - one performs check in Linux, the other one makes changes in the firmware, and no checks in Linux; perhaps this is intentional as changes with DCU cannot be made when locks are enabled, but I think this needs better explanation

2. https://github.com/Dasharo/open-source-firmware-validation/commit/112a434c9d9e26443439ad93c4f15258762b3606
  - you remove this keyword from UEFI lib
  - looks like unintentional change, needs confirmation

filipleple, 2024-11-07:
@macpijan 
1. Yes, this was intentional, that seemed to be the fastest way to make sure that a suite follows through only with flash locks disabled, regardless of whether it's a laptop or a serial access platform. That should be properly documented though, which I've omitted - will fix that if we decide to bring the keyword back
1. Can't speak for this one, @philipandag?

philipandag, 2024-11-07:
This commit was merged as a part of https://github.com/Dasharo/open-source-firmware-validation/pull/399 which is completely unrelated to the keyword. It must be an unfortunate mistake that was overlooked by me and the reviewer and should be restored.

JanPrusinowski, 2024-11-12:
My PR resolves the removal of this KWD and also adds missing info about flash die:
https://github.com/Dasharo/open-source-firmware-validation/pull/493

macpijan, 2024-11-14:
> Yes, this was intentional, that seemed to be the fastest way to make sure that a suite follows through only with flash locks disabled, regardless of whether it's a laptop or a serial access platform. That should be properly documented though, which I've omitted - will fix if we decide to bring the keyword back

@filipleple yes, please do that.

I will finalize the https://github.com/Dasharo/open-source-firmware-validation/pull/493/files with second point, and have it merged to `develop` first.

filipleple, 2024-11-19:
[Added](https://github.com/Dasharo/open-source-firmware-validation/pull/589) the documentation