Dasharo/open-source-firmware-validation #251: APU: TPM tests should properly shutdown the DUT

pkubaj, 2024-03-28:
### Device

APU2/3/4/6

### RTE version

1.1.0

### Brief summary

TPM tests currently don't shutdown the machine after test

### How to reproduce

Run a couple of TPM tests.

### Expected behavior

All the tests should pass.

### Actual behavior

After a couple of tests, next tests fail but pass when executed manually.

### Link to screenshots or logs

[dasharo_security.tar.gz](https://github.com/Dasharo/open-source-firmware-validation/files/14793589/dasharo_security.tar.gz)

macpijan, 2024-05-10:
@krystian-hebel the proper shutdown should happen in TPM tests, or in all tests, so this problem does not occur?

krystian-hebel, 2024-05-10:
At least in test that runs right before TPM tests may be enough.

krystian-hebel, 2024-08-07:
Suite setup could check if the TPM timer reports as secure, and if not, keep the platform running until it does (which takes up to 2^22 ms, ~70 minutes), reboot properly and then start testing.