Dasharo/open-source-firmware-validation #253: No Get Sign Of Life keyword breaks DMI004.001 on APU

pkubaj, 2024-03-29:
### Device

APU2

### RTE version

1.1.0

### Brief summary

No keyword with name 'Get Sign Of Life' found. Did you try using keyword 'RequestsLibrary.GET' and forgot to use enough whitespace between keyword and arguments?

### How to reproduce

`robot -b command_log.txt -v snipeit:no -L TRACE -v rte_ip:192.168.10.172 -v device_ip:192.168.10.36 -v config:pcengines-apu2 -t DMI004.001* dasharo-compatibility/dmidecode.robot`

### Expected behavior

Test should pass.

### Actual behavior

`No keyword with name 'Get Sign Of Life' found. Did you try using keyword 'RequestsLibrary.GET' and forgot to use enough whitespace between keyword and arguments?
`

### Link to screenshots or logs

[command_log.txt](https://github.com/Dasharo/open-source-firmware-validation/files/14806152/command_log.txt)

macpijan, 2024-04-09:
@PLangowski I propose we drop the `RELEASE_DATE_FROM_SO`L variable, along with the `Firmware Release Date Verification From SOL` keyword.

PLangowski, 2024-04-09:
@macpijan ok, I'll do that

closed 2024-04-09 by macpijan